drspangle/infidel-sensor #4: Improve Calibration Procedure - User Feedback

drspangle, 2021-02-23:
Right now, the calibration procedure totally opaque with essentially no user feedback. It would be great if the calibration procedure for the sensor incorporated flashing LEDs to alert the user to the current state of the calibration process. There should be some sort of way to signal the following states:

1. Calibration in progress, insert pin
2. Reinsert pin/retry
3. Calibration stage success, insert next pin
4. Error

Each of the "flash codes" for these states should be documented and the calibration procedure revised.

blurfl, 2021-02-28:
How about adding an I2C display?

drspangle, 2021-02-28:
Nothing wrong with this idea, but it could potentially raise the BOM cost significantly. At the very least, it will require changes to the PCB design and layout. Since calibration shouldn't be happening very often, it might be fine to stick with flash codes for now, since there are only 4 states.

drspangle, 2021-03-05:
It looks like in #8 there was an I2C interface implemented, though I'm not sure how it works and haven't tested it yet. Perhaps there is some potential to do this calibration by connecting to a remote serial console and doing the calibration interactively through that.

blurfl, 2021-03-05:
#8 does set up a serial user interface on the Arduino host, with a 2-way communication to the ATTiny85.