drspangle/infidel-sensor #7: Which TinyWireS library to use

blurfl, 2021-03-01:
calibration.ino and driver.ino both use calls to TinyWireS.write() to send data, but I haven't found a version of the TinyWireS library that includes a write() function instead of send(). Is there a particular library version needed here? Changing .write to .send looks proper and compiles, but I'm not able to test that it works properly.

drspangle, 2021-03-02:
I can't say for sure what the correct library is here as I've yet to construct a prototype for testing myself. If you can show that this works with that change, I strongly encourage a PR which I'll accept expeditiously.

midopple, 2021-03-05:
Hi, 
i use this TinyWireS libary

https://github.com/nadavmatalon/TinyWireS

drspangle, 2021-03-05:
@blurfl is the library which @midopple linked to the same as the one you attempted to use?

blurfl, 2021-03-05:
Well found, @midopple. I had not seen this library, but it seems very complete and uses the Arduino Write library's write() approach used by the Arduino host (see Host_ee_prog.ino in #8 for instance). It uses the MIT license, not sure how that meshes with the rest of the project - IANAL.
There would be a benefit to adding the library you settle on to this project 😉

drspangle, 2021-04-24:
It seems like we've settled on the one that @midopple is using, at least in the alternate firmware he's got in here. I think this can be closed now.

closed 2021-04-24 by drspangle