drspangle/infidel-sensor #11: The pick and place locations are incorrect in the current CPL file

darraghbr, 2021-03-07:
If you go through the order process with JLCPCB the resulting pick and place locations are incorrect.

[image]

drspangle, 2021-03-07:
So is the issue that the pad placement for the IC3 chip and the other SMD component is incorrect, or is there a deeper issue here? I'm not seeing precisely what the issue is. Does this screenshot come from JLCPCB's design checker?

darraghbr, 2021-03-07:
I think it's a JLC issue, I've checked and double checked the positions and it just seems the presence of the switch in the CPL file breaks everything.

blurfl, 2021-03-07:
We shouldn't modify our BOM to work around one vendor's internal issue. They correct for it when they review the order.

midopple, 2021-03-08:
I had the same problem with JLCPCB, after contacting support I was told that this is due to the web preview and is still being checked before the production process.

electrickery, 2021-03-20:
My boards arrives some days ago and the components are all at the right places. Haven't started and programmed the board yet.

coelholm, 2021-04-09:
I saw same issue. It is an issue with JLCPCB web preview. I have tried to correct placement position of each component one by one (It was looking fine), but when I got to the last component, everything looked "misaligned" again, so I decided to use original file and my boards are perfect.

drspangle, 2021-04-16:
It seems like this isn't actually an issue, rather is a problem on JLCPCB's end. I'll close this now.

closed 2021-04-16 by drspangle