drspangle/infidel-sensor #14: No mention of spring pre tensioning hardware

darraghbr, 2021-03-07:
I believe there is a missing item on the hardware BOM, I believe there should be an M5 grub screw or the like to pre load the spring in the body of the sensor.

drspangle, 2021-03-07:
I believe you are correct. Tom originally specified that he used a grub screw in his video but I don't know what was specified. Can you verify that an M5 grub screw works, and if so, can you recommend/specify a specific one?

darraghbr, 2021-03-07:
> 
> 
> I believe you are correct. Tom originally specified that he used a grub screw in his video but I don't know what was specified. Can you verify that an M5 grub screw works, and if so, can you recommend/specify a specific one?

An M5 grub screw definitely does work, and either of the standard sizes (5 or 8 mm long) will work fine.

drspangle, 2021-03-07:
Great! This should definitely get added to the BOM. If you want to create a PR to add this I will review and accept it expeditiously. Nice catch on this one.

drspangle, 2021-03-07:
Fixed in #16.

closed 2021-03-07 by drspangle