drspangle/infidel-sensor #9: The lever part has some thin walls that make it almost unprintable #4

darraghbr, 2021-03-07:
The walls that the bearings locate to are extraordinarily thin, so thin in fact that they are basically unprintable with a normal FDM printer with a 0.4mm nozzle. Even with a 0.25mm nozzle I don't think these details are resolvable in a suitable material ASA, PETG as specified by the original designer.

image
[image]

Whereas the solid part should look like:

image
[image]

Looking in the CAD the inner and outer faces appear to be intersecting, which will absolutely make any printer sad, as it's trying to print something that is infinitely thin.

drspangle, 2021-03-07:
Is there any likelihood that this was done on purpose?

deezums, 2021-03-24:
The closer the fulcrum is to the fixed pivot the larger the movement at the end of the lever, in other words it's probably on purpose. Nothing about that is unprintable, nor would it be mechanically unsound.  Both bearings and the axle have plenty of material constraining them.

darraghbr, 2021-03-24:
That is not at all what we are talking about or what I have changed in the linked PR.

drspangle, 2021-03-24:
@darraghbr I apologize for the confusion, could you help everyone understand a little better what you are trying to address, versus what @deezums has raised?

darraghbr, 2021-03-24:
In the original CAD there is an infinitely thin wall behind the upper part of the bearing seat, which no printer will be able to produce, leading to the enormous holes that you seen in the preview at the top of this issue. So the first thing I did was make it so there is actual material that sits behind the bearing. 

The other thing I did was reduce the chamfer on the hole for the bearing, because as you can see in the original CAD the chamfer intersects with the hole for the axle pin and there is an increased likelihood during assembly of the axle breaking out through the thin wall and rendering the print useless.

drspangle, 2021-03-24:
@darraghbr Okay, thanks for explaining that.

So it seems there are two issues addressed here, the first being a cosmetic issue that affects slicers but perhaps less dramatic of an effect on the overall print. The second issue is improving the fitment for the axle. There should not be much of an effect on the overall positioning of the axle or the attached lever arm from these changes. Have I got that right?

darraghbr, 2021-03-25:
The first issue is not just cosmetic, it gives the bearing a better seat and means the bearing is supported all the way around it's circumference, rather than just the lower 2/3rds.

drspangle, 2021-04-16:
This has been resolved in #10.

closed 2021-04-16 by drspangle