drspangle/infidel-sensor #18: Add Electronics BOM

lost-hope, 2021-03-24:
Hello,

would people be interested in electronic BOMs for both PCB versions?
I made a BOM for my build and could add it here and maybe add links to electronic resellers like mouser or similar (EU and US)

drspangle, 2021-03-24:
I assume you're referring to your fork based on pi nano? If that's correct, then I think there are some tough questions that need to be considered about how we can properly version both of these simultaneously, and it may be more suitable to have the pi nano as a separate project, or to pull in limited revisions from a particular branch of the main project.

If you're talking about something else, some more details would be great. In any case, having links to recommended resellers serving a variety of geographic areas would be great; EU (not sure what to recommend here, Amazon perhaps?), North America (Mouser, DigiKey, Amazon, etc.), and Asia (AliExpress) ideally.

blurfl, 2021-03-24:
Having a text BOM for the thru-hole version as well as one for the smt version that is separate from  'electrical/smt/gerber/filasense_top_bom.csv' would be a good thing. That file is in a format specific to ordering from JLCPCB, and I think it should be left as is. 
 Better still might be to roll the BOMs and the schematics into an .md file to provide an overview independent of Eagle CAD.

I'me the guilty party who mentioned the rpi pico fork, it really does belong in a separate project or branch.

lost-hope, 2021-03-24:
I have started working on a BOM as a MD file. Currently I included Mouser as a vendor, cause they cover EU and US.
https://github.com/lost-hope/infidel-sensor/blob/main/electrical/smt/BOM.md
@drspangle this is what i mean. A table with the electrical materials listed.

@blurfl what do you want to do with the rpi pico? having a standalone sensor?

blurfl, 2021-03-24:
`rpi pico? having a standalone sensor?`
That, and a cheaper build cost. The power supply seems _very_ noisy, though.

darigovresearch, 2021-04-10:
[Kitspace](https://kitspace.org/) may be the best user friendly implementation of a BOM to resolve the sourcing of electrical components of this project. May be worth checking which suppliers are available and crowd sourcing from the users here who know the catalogues of those suppliers best. Would suggest an initial BOM to be confirmed from one source of parts if the maintainers choose to implement it.

See feature request #19 for more details

drspangle, 2021-04-16:
@lost-hope If you want to make a PR to incorporate your BOM, please do so and I'll make sure it gets reviewed.

lost-hope, 2021-04-16:
I opened PR #20 with the current state and only one vendor (Mouser), cause they seem to ship in both US and EU. Though the shipping cost are quite hight

darigovresearch, 2021-04-17:
I think the pull request resolves this issue, separate issues can be opened for the other vendors to see if anyone else in the community wishes to contribute part numbers for those vendors.

drspangle, 2021-04-24:
This seems to be addressed adequately for now. We'll track the addition of other suppliers in #32 .

closed 2021-04-24 by drspangle