drspangle/infidel-sensor #19: [FEATURE REQUEST] Add Kitspace one click pricing

darigovresearch, 2021-04-10:
[Kitspace](https://kitspace.org/) is a project that really helps with building open source hardware as it allows you to (when properly implemented) have an almost one click pricing for the PCB and electronics components. If setup by the maintainers it would really decrease uncertainty if the components chosen are the ones that work reliably.

We haven't had to implement it yet but it seems relatively straightforward to implement and lets users have self-autonomy to replicate the work, build upon it and put it in the hands of users.

Examples of very good implementations of this are the [Open book](https://kitspace.org/boards/github.com/joeycastillo/the-open-book/open-book/) and [ebook wing](https://kitspace.org/boards/github.com/joeycastillo/the-open-book/ebook-wing/) projects by @joeycastillo

https://kitspace.org/1-click-bom/
https://github.com/kitspace/1clickBOM#readme

Would also resolve issue #18 as well

drspangle, 2021-04-16:
What is required to do this?

drspangle, 2021-04-16:
@darigovresearch If you want to set this up and create a PR, I will review it.

darigovresearch, 2021-04-17:
Thanks @drspangle, pull request #23 should contain the files that are required for this repository

drspangle, 2021-04-18:
Addressed in #23.

closed 2021-04-18 by drspangle

kasbah, 2021-04-19:
It's up! https://kitspace.org/boards/github.com/drspangle/infidel-sensor/