drspangle/infidel-sensor #24: [FEATURE REQUEST] KiCad version of PCB

darigovresearch, 2021-04-17:
We may get more contributions/improvements to the system if it is in an open source PCB format like KiCad as it would reduce the barrier to entry for new users. Was wondering if the maintainers or someone from the community would be interested in trying this out.

https://www.kicad.org/

midopple, 2021-04-17:
I have add the KiCad Files for the SMT Version, see Pul request #26

darigovresearch, 2021-04-17:
@midopple thanks for taking the time to do this, have managed to see if we can open the files & it seems to be working. Unsure if the porting has resulted in any differences from the original as we're still relatively new to PCB design & KiCad

[image]

drspangle, 2021-04-18:
I can't tell if there are any differences, and this will potentially create issues if both sets of schematics are not updated, but I'm of the opinion that if this helps the project gain traction among the community then it's worth the risk. Addressed in #26.

closed 2021-04-18 by drspangle

darigovresearch, 2021-04-24:
@midopple thanks for the pull request & @drspangle thanks for merging it. We also believe that having both copies is worth the risk if it encourages people to join the project. Keeping it updated in KiCad could also be good easy pull requests for people who want to get started in contributing to KiCad projects.