drspangle/infidel-sensor #27: Capacitor grounds not connected

eltobu, 2021-04-23:
As can be seen in the screenshots below, the grounds of the two capacitors are connected to each other and a ground fill island on the bottom side.
This ground fill island is then not connected to the rest of ground, but it should be.
Top side:
[image]
Bottom side:
[image]

blurfl, 2021-04-23:
That's a good find. Could you add pics to PR #28 to show the fix?

eltobu, 2021-04-23:
Yeah, would have been a good idea in the first place.
I have added screenshots of the full board and both layers.

closed 2021-04-23 by drspangle

darigovresearch, 2021-04-24:
@eltobu good catch, & @blurfl thanks for the fix. Can anyone tell if this issue is also in the KiCad version of the project as it was autogenerated a week before this fix?

drspangle, 2021-04-24:
@darigovresearch As far as I can tell, #28 addressed the problem in the KiCad version, and #29 addressed the problem in the EAGLE version.

darigovresearch, 2021-05-09:
Thanks @drspangle for confirming & @blurfl for the fix!

PR #28 (2021-04-23): fix capacitor grounds not connected
Fixes #27 in KiCAD only