drspangle/infidel-sensor #32: [FEATURE REQUEST] Add remaining Kitspace distributors for electronic components

darigovresearch, 2021-04-24:
As a [Kitspace page](https://kitspace.org/boards/github.com/drspangle/infidel-sensor/) has now been implemented to automatically order parts & get pricing it would be good to get any of the remaining suppliers in order to find best prices. If anyone in the community knows any of these catalogues well it would help get this design into the hands of users to help test and improve the project.

Happy to split this into individual issues to help encourage contributors if the maintainers agree as then it is easier to assign specific retailers to specific users and use this one to track progress overall

Suppliers implemented
- [x] Mouser

Suppliers not yet implemented
- [ ] Digikey
- [ ] Farnell/Element14
- [ ] Newark
- [ ] RS
- [ ] LCSC

drspangle, 2021-04-24:
This should be fine as a single issue. I'll have you be in charge of this one.

darigovresearch, 2021-04-24:
Happy to direct contributors on how it works but may not be sure if the components selected should be vetted by more experienced people with this project to avoid anyone (particularly those new to PCB design) ordering incorrect parts and then being turned off from the project when it doesn't work as expected. 

May be good to add the tag `Good first issue` and `Help wanted` tags to this issue to encourage people to take part if possible