drspangle/infidel-sensor #36: issue with AISLER PCB ordering

nlooije, 2021-05-03:
When ordering PCB on AISLER (accessed through the kitspace website) the PCB apparantly isn't loaded properly:

- PCB, Stencil tabs don't load, are in infinite loop getting production date
- Parts aren't loaded from the BOM

This issue is solved by downloading the Github repo and manually uploading the EAGLE file on AISLER

drspangle, 2021-05-03:
Paging @darigovresearch ...

darigovresearch, 2021-05-03:
@kasbah is this a known issue for AISLER?

kasbah, 2021-05-03:
Thanks for reporting and ating me. Seems to be the same if you download the Gerber zip and upload manually. The Gerbers look OK except the outline doesn't show with Tracespace ([link](https://tracespace.io/view/?boardUrl=https://kitspace.org/boards/github.com/drspangle/infidel-sensor/infidel-sensor-60b6436-gerbers.zip)). The outline does show up when using Gerbv. There also seems some copper outside the outline, but I feel like this should still be able to work. 

Maybe @PatrickFranken has some insight why the Aisler preview can't cope with the Gerbers ([download here](https://kitspace.org/boards/github.com/drspangle/infidel-sensor/infidel-sensor-60b6436-gerbers.zip)) but can cope with the Eagle file.

kasbah, 2021-05-03:
Just a note: different gerber viewers render the outline differently. 

E.g. OSHPark
[image]

vs PCBWay (which looks very similar to how Kitspace renders it, they may be using Tracespace as well)
[image]

PatrickFranken, 2021-05-04:
It looks like this is not a general issue but rather limited to this project. To validate this, I double checked the Gerber files for this project. Please note that our Gerber preview is a bit different to other Gerber viewers as it displays the actual processed manufacturing files compared to others which just render the Gerber files. This also makes it a bit more "critical".
In this case the issue is caused by an indistinguishable, double drawn board outline, see screenshot attached. This makes it impossible to resolve *the correct* milling path for this PCB. As soon as this issue is solved the import should work as expected.
[image]

darigovresearch, 2021-05-08:
Good catch @PatrickFranken, is there anyone from the maintainers of the repository or the community able to resolve this issue in the eagle & KiCad files?

CCMCAGP, 2021-06-14:
Hi,
Related to this issue, I ordered via Aisler by uploading the EAGLE file manually, which gave this rendering:
[image]
and I sent the design, but the tests post-manufacturing indicate a big number of opens: 
[image]
 Can anyone indicate to the newbie that I am whether this is expected or not?  
 Thank you!

PatrickFranken, 2021-06-14:
We are currently seeing quite a lot of false-positive open connections after electrical test of the PCBs. I assume this is also the case here. As the PCB itself is rather simple I expect the manufacturing to be done properly.