drspangle/infidel-sensor #37: [FEATURE REQUEST] Documentation

darigovresearch, 2021-05-09:
Would be good to have a docs page to help improve adoption of the project such as step-by-step assembly instructions for the build and coding. This would help make the project more accessible for new joiners and help pull from a wider audience for new features etc. Would suggest having it in the same repository if possible so it could be hosted as a GitHub page

Example docs frameworks that can be used:
- Read the docs - https://readthedocs.org/
- Docusaurus - https://docusaurus.io/
- docsy - https://www.docsy.dev/docs/

Example of open source documentation based on some of the above frameworks
Open source pick and place machine - https://docs.indexmachines.io/docs/
Open source machines to recycle plastic - https://community.preciousplastic.com/academy/intro.html

Happy to write content for it if the maintainers think this would be helpful and choose which documentation framework they have a preference for

darigovresearch, 2021-05-15:
Sure happy to have a go, do you @drspangle or any of the other maintainers have any preference on the documentation framework?

drspangle, 2021-05-15:
Nope. I own the repo, and I'm open to anything.

darigovresearch, 2021-05-15:
Sounds good, an initial test build can be found in #38

drspangle, 2021-06-19:
I had to move the documentation into the correct folder so that the github pages URL would work properly, otherwise you get a 404 unless you explicitly add `/html/build/` subfolder in the URL.

drspangle, 2021-06-19:
Still doesn't seem to be working properly, all the static assets' locations don't seem to be defined correctly anymore.

drspangle, 2021-06-19:
Fixed, I forgot to copy over additional files that were required from the sources directory. Documentation is up and looks good. It would be nice if we had a way of automating the generation using jekyll but it's not a huge deal to just update the HTML every time we update the docs.

closed 2021-06-19 by drspangle

darigovresearch, 2021-06-19:
Yeah as you wish in terms of moving it to adjust the URL to not need to `/build/html/`.

In terms of automation, have seen on other repositories that it can auto build the docs for the pull request so it can be demoed before merging but also auto building the HTML as well. Have yet to try this out but may be a good new feature request to open