drspangle/infidel-sensor #39: License Query

darigovresearch, 2021-05-16:
In the readme it is stated that the project is under a CC0 design but it also links to a previous project which is under a different license (CC-BY-NC-SA) - https://www.youmagine.com/designs/filament-diameter-sensor

As it is derived from a project with a different license should the source project not also be changed for the CC0 of this project to be valid?

drspangle, 2021-05-16:
Thomas Sanladerer released this under CC0 and I chose to retain that licensing scheme. Refer to the original Youtube video that's in the documentation. I believe Youmagine doesn't support CC0 licensing, but if it does then that should be fixed there.

closed 2021-05-16 by drspangle

darigovresearch, 2021-05-16:
Thanks @drspangle for the clarification. Would agree that Thomas Sanladerer released this design under a CC0 from the video and knowing his work in open source he genuinely meant it to be under a CC0. Youmagine does handle CC0 licensing as found [this version of the design](https://www.youmagine.com/designs/infidel-inline-filament-diameter-estimator-lowcost-10-24) in the timeline of the project linked above. Would assume that as the previous project was released 7 years ago the new license hadn't been updated in the original project to match this derivative work.