drspangle/infidel-sensor #41: [FEATURE] Add firmware instructions to documentation

darigovresearch, 2021-06-19:
**Is your feature request related to a problem? Please describe.**
There currently is no instructions/documentation on the firmware as in the folder readme

**Describe the solution you'd like**
Take the content from [here](https://github.com/drspangle/infidel-sensor/tree/main/firmware) and put it in a page of the [documentation](https://drspangle.github.io/infidel-sensor/index.html)

**Describe alternatives you've considered**
Leave it in the readme but having it all in one place would be helpful

**Additional context**
N/A

leonhma, 2021-07-11:
made #54 to reflect latest changes

darigovresearch, 2021-08-08:
Thanks @leonhma, hopefully the discussion in https://github.com/drspangle/infidel-sensor/pull/48 will help move this forwards

leonhma, 2021-08-20:
I made #54 by just converting `firmware.md` and adding it online. (Still has to be merged I think). As #55 noticed, some details don't match up. I think we should merge it first and then figure out some of the wrong details.