drspangle/infidel-sensor #44: [FEATURE] Add auto-build of documentation as preview in a Pull request

darigovresearch, 2021-06-19:
**Is your feature request related to a problem? Please describe.**
When making a pull request aiming to improve the documentation, it is hard to test that the new changes does not break the original documentation 

**Describe the solution you'd like**
Figure out a way to auto-build it in a check when merging a pull request

**Describe alternatives you've considered**
Manually building but this takes time

**Additional context**
Example for this feature using readthedocs can be found in the [Manim Community repository](https://github.com/ManimCommunity/manim/)

Example pull request - https://github.com/ManimCommunity/manim/pull/1712
Example build for the pull request (may not last after merging) - https://manimce--1712.org.readthedocs.build/en/1712/

darigovresearch, 2021-06-19:
@Darylgolden do you know how this was done on the Manim Community repository?

WampyCakes, 2021-06-20:
Are you referring to [this RTD setting](https://docs.readthedocs.io/en/stable/pull-requests.html)?

leonhma, 2021-07-10:
You could actually scrap the Workflow Part i Just added in #52 i think, move the Docs to rtd and enable this setting.

darigovresearch, 2021-08-08:
@leonhma would be very happy to see this feature implemented if you're still up for doing it

leonhma, 2021-08-09:
Yeah sure but I don't think I'm maintainer yet. (I missed the invite 😬)

leonhma, 2021-08-09:
Although, it would be better if you @drspangle set up the project on RTD (I don't know if I can add the RTD oath app even as maintainer) setting up the project is quite easy 👍

drspangle, 2021-08-09:
@leonhma You have the ability to add this to your RTD account -- you just need to add your GitHub account to RTD and then you can authorize the application to pull in the repo from GitHub, not the other way around.

If you already sign in with RTD using your GitHub account, you're done.

https://docs.readthedocs.io/en/stable/connected-accounts.html

leonhma, 2021-08-09:
yeah but... it also needs this 'oauth app' or whatever that automatically builds on pull requests and i dont think you can do that as a maintainer? i can try though..

leonhma, 2021-08-09:
https://infidel-sensor.readthedocs.io/en/latest/ here it is. it tells me that i dont have the permissions to set up a webhook, though.

drspangle, 2021-08-09:
@leonhma I can set up the webhook. You will need to send me the payload URL and secret via a secure medium -- you can find me on Keybase at the same username as my GitHub account.

leonhma, 2021-08-20:
@drspangle I think you can close this, should be resolved via the setting in RTD

drspangle, 2021-08-20:
Agreed. Closing now.

closed 2021-08-20 by drspangle