drspangle/infidel-sensor #50: [FEATURE] 2.85mm version

Titi9237, 2021-06-26:
Is there a plan for a 2.85mm version of this sensor ?

drspangle, 2021-06-27:
At the moment, we don't have plans for this. However, there's no reason why it couldn't be created with a few modifications to the 3d printed parts.

darigovresearch, 2021-08-08:
@Titi9237 would you be up for making the adaptations for a 2.85 mm version and making a pull request?

Titi9237, 2021-08-08:
I know what modification are needed and it is on my to-do list but unfortunately I am mainly using FreeCad (Linux) so I need to find a Windows computer to make a "clean" 3D file.

Titi9237, 2021-12-23:
I finally found time to do a very rough job to make it work with 2.85 filament. 
Basically, expanding the filament path by 1mm in radius to account for the extra diameter of 2.85 filament. Without major changes, there is no way to include a corresponding PTFE tube. We need two bearings that have a 9mm outer diameter (MR93ZZ) in place of the 10mm outer diameter previously used (625ZZ).

The parts will be printed during the holiday to see if the mechanical dimensions are correct. Adn then I will to order the remaining parts to see if everything is ok.