drspangle/infidel-sensor #55: Make the firmware documentation more explicit

CCMCAGP, 2021-08-19:
I would like to make the doc&gt;firmware.rst document more complete and accessible. Can
someone review it and see whether my initial steps are correct?
As a beginner in electronics, I have been struggling to interprete the firmware instructions, but a research project I’m doing requires this filament sensor so I'll have to make it work, on a printer too!

> ### REQUIREMENTS 

-  Arduino > V1.8.8
-  TinyWireS LIbary from https://github.com/nadavmatalon/TinyWireS **Download the .ZIP of this library and upload in Tool>Include Library>Add .ZIP library**
-  ATtiny85 Board for Arduino from https://github.com/damellis/attiny **Load this in Board Manager, more info in** https://www.youtube.com/watch?v=AmpHIHM41Hw&ab_channel=Adio_ -
-   ISP Programmer like Arduino over ISP -->
   https://www.arduino.cc/en/Tutorial/BuiltInExamples/ArduinoISP **Set the Arduino Uno as an Arduino As ISP following same video as above. Upload the example ketch ArduinoISP onto the board, this board is now an ISP programmer and you may need a different board as host.** 

> Load the code, wire the ISP program to the ISP port on the sensor and
> program it.

This sentence was very obscure to me, so here's how I interpreted and detailed it.

## How to flash firmware onto the sensor 

1. Wire the ISP programmer to the ISP port on the sensor.
I think this is different from the host_to_sensor_arduino.PNG wiring drawing because the ISP port is not occupied. So I tried to find the equivalent functions of each pin  on the ISP port and on the Uno ISP programmer.

Type | Pin #   on ISP  port (Sensor) | Arduino Pin (ISP programmer)
-- | -- | --
MISO | 1 | 12
VCC | 2 | 5V
SCK | 3 | 13
SDA | 4 | 11
RESET | 5 | 10
GND | 6 | GND

2.	In Tools, Set processor to Attiny85, Clock to Internal 16MHz **(current doc says 1MHz but the sensor .ino files all say 16MHz),** Programmer to Arduino as ISP
3.	Tools > Burn Bootloader
4.	Open driver.ino and Sketch>Upload with Programmer 

## Current status
At the moment, with this method, three issues arise: 
**ISSUE 1.** In the doc Firmware, the LED is said to flash twice. For me, it flashed only once. 
**ISSUE 2.** With the driver.ino flashed onto the sensor, when I approach a magnet, no matter what the magnet polarity is, the LED is triggered. I wonder if anyone has the same issue. In the video https://www.youtube.com/watch?v=RYgdLPe_T0c&t=14s&ab_channel=ThomasSanladerer (5:19), the sensor should not be triggered with one of the two magnet faces (and that's the direction you should use to stick on the lever). I have a N45 magnet instead of the N35, but I also have the same problem with a much smaller N45 magnet which should have a smaller force. 

**ISSUE 3.**  With the current state of the documentation, I don't understand when the 5 different firmwares should be used. Could someone clarify the difference between all these firmware in the description and state exactly on which occasion they should be used? 

**TLDR;** 

a noob tries to clarify the documentation on the firmware. Can you review up to this stage, and clarify what firmware should be on host and sensor at every step?
 

PS: I realize this might belongs more to a forum than to GitHub but since I'm not aware of one, please forgive me for posting it here. I hope that maybe my remarks and questions could help to make this project more accessible to others as well.

drspangle, 2021-08-19:
This is a good a venue as any for comments and feedback. I'll take a moment later on today to review this and I'll try and help you out. Other contributors might beat me to it.

leonhma, 2021-08-20:
Doesn't look wrong to me. I can't test it though, my parts are still stuck in China 😬. As you saw, I made a pr (ran firmware.md through a converter and added it to the online docs). I don't really know how detailed this is. @CCMCAGP  you can use the MD document in the `firmware` folder as reference.

CCMCAGP, 2021-08-20:
Thank you for your answers. I realized what is wrong: at this point it is not driver.ino but calibration.ino that should be loaded onto the sensor. Even better to follow along with the firmware.rst doc, these two files are actually not useful as far as I know. To start calibration, you should upload Host_ee_prog.ino onto the Host and Infidel_release_ee onto the sensor. Then you start seeing the next steps according to firmware.rst. 
When I'm through with the calibration I will post a more thorough summary!

leonhma, 2021-08-24:
Looks good to me. Feel free to open a PR to my repo and then it can be merged back into this one.