drspangle/infidel-sensor #60: [Bug]: Link to docs not the latest

darigovresearch, 2021-08-29:
### What happened?

Link in the readme points to the old GitHub pages sphinx documentation

### Any solutions?

Put in a new link  now that it is in readthedocs-> https://infidel-sensor.readthedocs.io/

drspangle, 2021-11-26:
It seems like #61 broke something -- all of the styling/sphinx theme for the docs appear to be missing on the version at https://infidel-sensor.readthedocs.io/

@darigovresearch Can you confirm?
@leonhma I see that you updated the webhooks for this, is it configured correctly?

leonhma, 2021-11-26:
@drspangle can confirm. I cloned the repo locally and tried stuff, and it seems like the table-scrolling-fix in conf.py ist unreliable and causes the build to fail. You can just remove the last 5 lines of the conf.py file and everything should work again.

closed 2021-11-26 by drspangle

darigovresearch, 2021-11-26:
@leonhma nice catch, it appears @drspangle done your suggestion & it seems to have resolved everything now on the linked page - https://infidel-sensor.readthedocs.io/en/latest/index.html

commit 7717a6b1 (2021-11-26): Update conf.py

Removed the code which @leonhma recommended, in order to fix bugs introduced in #61. Should fix #60.