drspangle/infidel-sensor #62: [Feature] Filament stuck function

LordPinhead, 2021-12-16:
### What's your feature?

Could you add a function with an high resolution encoder wheel, to know when filament is stuck? Bigtreetech has something like this with their sensor, but it's really not that reliable:

https://www.youtube.com/watch?v=UUx2R9u6ohE 

This review of this sensor shows the problem. If you could add a version, in which one of the bearings is the encoder wheel and add that to the sensor, that would be really a gamechanger. When I look at what you did so far, that can't be that hard.

### Additional context?

None

drspangle, 2021-12-16:
Actually you'd think this is a very simple feature to introduce but it's
shockingly difficult. A primitive version would be to check that the
variance of the filament has not changed beyond a very very small
threshold, but then you'll get false positives when the filament is
actually just very precise. Also, when jamming occurs, the vibration
induced along the filament path from clicking steppers could potentially be
detected as changes in variance. This is a tough problem. Does anyone have
any suggestions that would not require a total redesign or the introduction
of an additional or different sensor? An encoder wheel would do this very
easily for example but that is a  substantial redesign.

LordPinhead, 2021-12-17:
I don't know if you watched the video, this sensor is working somewhat decently, but it has its flaws. But when we think of the problem, what options do we have:

1) Optical sensor: Won't work because the filament would need to have changes in the structure. Optical mice work like that.
2) Mechanical wheel / encoder wheel: Depending on the resolution, this could work. But when you have to completely redo it, that would be not good. We would like to keep the current design and add the functionality.
3) Knocking sensor: Like in a car, the sensor could be on the extruder and fetches "knocking", it should be fine enough for the stepper motors we have to get the jumping of the gears and send it to the board.